cabeese/crab-tracker #2: Statistical analysis - likelihood of collisions

cabeese, 2017-11-04:
We want to know the statistical likelihood of transmission collisions given various crab densities, broadcast encoding lengths, and interval lengths. We'd like to know how varying any one of these values affects the expected number of collisions within a given period of time.

chloeyugawa, 2017-11-29:
@cabeese Please add some sample numbers (e.g. interval time, etc.) so we can have some more-realistic calculations.

cabeese, 2017-12-14:
Here are some numbers that may be helpful and more in line with what we use in the project. Please forgive my shorthand notation.

### Rough estimates
* The **target listening radius** (aka the "**detection radius**") we're shooting for is about 50m.
* The **length of a signal**, from the rising edge of the first ping to the falling edge of the second ping, _is very much subject to change_ as the project progresses. However, for the time being, let's use the following numbers (derived below): 11ms to 1511ms.
* The **interval time**, defined as the time between consecutive broadcasts by a single transmitter*, is a randomly-varying value that will be recalculated between each broadcast. Currently, that value is expected to range between 20 and 30 seconds. _This, too, is subject to change_.

### Calculation of signal length
Pings should be a minimum of 1ms, and the space between rising edges must be no less than 10ms. The "step size" for ID encoding could be 0.1ms (see the equation below).

We'll assume `ping_duration = (1 + ID)ms` and `delay = (10 + 2*ID)ms`. Delay is calculated between rising edges, so one signal's length would be `ping + (delay-ping) + ping`, or just `ping + delay`. Therefore the min length of the whole signal would be `ping(0) + delay(0) = 1ms + 10ms = 11ms`, and the max (assuming a max id of 500) would be `ping(500) + delay(500) = 1ms + 500ms + 10ms + 2(500)ms = 1511ms`, or roughly 1.5 seconds.

### Footnotes
\* Footnote about the interval time: I think that this is defined as the time between the falling edge of one broadcast and the rising edge of the next, but it may be the time between the rising edges of two consecutive signals. However, for the sake of this document, I think it's acceptable to leave things ambiguous; the discrepancy in 'interval length' values between the two definitions is only 1.5 seconds, as we can see above.

cabeese, 2018-01-19:
Based on our recent discussions, it seems like an average of 30 seconds is a bit high for the interval time. What might it be like if we changed that range to be 7 to 13 seconds? (Something I'm just throwing out off the top of my head right now.)

31337H4X0R, 2018-01-19:
Ultimately I think that the interval time delay will depend on the
detection range; a larger detection range will result in more potential
collisions, so we may need to make the interval time longer just to avoid
unreasonably high collision probability. That, in turn, depends on basic
hardware testing that *should* be completed in the near future. I'll update
you all on that as soon as John and I get there. Since sound transmission
underwater is essentially lossless, I think that it will be quite hard to
keep the detection range as small as I want. So, I expect that the
detection range will be quite a bit more than 50m (just a gut feeling).

In the ideal scenario with low detection ranges those interval times all
sound reasonable to me. After thinking about it a bit, I don't think there
will be any reason to have an interval time less than 10-15 seconds

patrick

cabeese, 2018-01-19:
Thanks for the input, Patrick! Great to hear your thoughts on that. To me, it sounds like our best bet for lowering the detection radius, should we need to, would be to decrease the power of the transmitters. Is that your feeling too? And how feasible is that with the hardware we're using? We could tune the integrators on the receivers to be less sensitive, but I think we could still run into issues with collisions from very distant transmitters that way. Those are just my gut feelings, too, though. I'm excited to see what we learn when we start testing all this stuff!

31337H4X0R, 2018-01-19:
I'm not sure whether adjusting the transmitter output or the triggering
threshold on the receiver would be better. I think we'll need to wait until
we get some data from prototype tests before we can move forward. I'm sure
John will have a more informative answer.

cabeese, 2018-01-21:
I realized I made some errors in those initial calculations. Specifically, I forgot a rather important decimal place. Here's some updated info which I shared with @chloeyugawa earlier.

Duration for each part of a UTP (see iCRAB Protocol Definition):
* `ping: (1 + 0.1*id) ms`
* `delay: (10 + 0.1*id) ms`
* `UTP: 2*ping(id) + delay(id)` (UTP is the full signal (ping, delay, ping))
* `inert_ping = (max_utp + delay(id) + ping(id)) ms` (method 4.3 only)

With these updated numbers, the upper bound on our UTP duration is much higher. If we encode boolean values via method 4.2 (see RFC 2), meaning that we use twice the number of IDs we expect to need and reserve half of them for "inert" crabs, our max broadcast duration will be **212 ms**. If we use method 4.3 (make the first ping really long if `inert = true`), the max broadcast duration is **323 ms**. These numbers assume that our biggest ID is 500 (which means we encode 1000 IDs in method 4.2).

31337H4X0R, 2018-01-21:
Did we ever discuss with John whether we can detect falling edges on the pings? We should bring this up in the meeting tomorrow (unless it's already been addressed and I'm just being forgetful).

Along the same lines, let's also run your idea for encoding the boolean by John tomorrow to keep him apprised of your thoughts on that.  Is it necessary to vary the length of both pings for collision detection? If we can get away with encoding ID redundancy in one ping, the second ping could be kept much shorter at all times and still be used to encode the boolean. This would decrease the probability of collisions by decreasing overall signal length.

cabeese, 2018-01-22:
I believe we confirmed that we can detect falling edges. @mimaxwell, can you confirm? In your `READ_PIND` code you were able to read everything, right? As I remember, you can read any pin at any time and it returns a `0` or a `1`, so we can detect a `1 -> 0` transition (aka a falling edge).

As far as varying the second ping length, I'd have to give that some more thought to be absolutely sure one way or another. I don't remember off the top of my head if we will run into issues if the second ping is a fixed length. However, at this point I'm not too worried about shaving off time from the UTPs. If transmissions are ~0.2 seconds and they occur every 10-20 seconds, for example, I think collisions are still going to be fairly infrequent given our density estimates. @chloeyugawa has been working on new statistics with some updated numbers, so we'll be able to quantify "fairly infrequent."

chloeyugawa, 2018-03-20:
Closed by #42

closed 2018-03-20 by chloeyugawa