cabeese/crab-tracker #13: Research unit testing in C/C++

cabeese, 2018-01-13:
# Background

Especially for components such as our directional vector algorithm, we want to ensure that our code performs as expected when given various input. However, none of us are familiar with standard "unit testing" methods for C/C++ development. We'd like to know common practices so that we can establish consistent testing methods across the project.

# Requirements

Figure out good or common methods for writing and running unit tests in C/C++ and document these somewhere (as comments to this issue, documents or samples in the repo, etc.).

enalaiya, 2018-01-19:
There are several good frameworks for this - we should just pick one.
Check or cUnit seem good!

cabeese, 2018-01-19:
Are there any striking differences between the two?

A couple of others that look promising:
* https://github.com/cpputest/cpputest Aimed at Embedded Systems devs, but it looks like a nice interface
* https://github.com/google/googletest New and actively developed by Google

cabeese, 2018-02-11:
Ohhh, this looks cool! https://github.com/catchorg/Catch2

Never seen it before. Looks lightweight and really easy to get into. I may start using it for my code.

cabeese, 2018-02-20:
I've made some unit tests with [Catch2](https://github.com/catchorg/Catch2) and I'm really liking it. I suggest that we move forward with it for the time being. I'm closing this issue for the sake of the deadline tonight, but we can revisit this topic later if we feel that Catch2 isn't able to provide us with everything we need.

closed 2018-02-20 by cabeese