cabeese/crab-tracker #37: Bring up Digital Compass

cabeese, 2018-02-23:
# Background

Collectively, we determined that a magnetic compass is necessary in order to make this program more usable. We have since purchased two such chips and we'd like to add headers and wire them up.

# TODOs

* [ ] Solder headers on both boards
* [ ] Create jumpers
* [ ] Connect the boards to the Pis
* [ ] Ensure that the screen and the compass are configured properly to be addressed individually over I2C - that is, both devices can be used at once without interfering with each other.

mimaxwell, 2018-03-11:
Needed to: 
sudo raspi-config
select option 5 Interfacing Options
go to P5 I2C and press enter
A screen will pop up that asks "Would you like the ARM I2C interface to be enabled?" select <Yes>
Select <Ok>
Select <Finish>

vim  /etc/modules
add the line i2c-bcm2708 

vim /boot/config.txt
uncomment or add the lines 
dtparam=i2c1=on
dtparam=i2c_arm=on

sudo reboot

i2cdetect -y 1

Currently the pi won't recognize the screen and the compass at the same time

cabeese, 2018-03-14:
@mimaxwell Looks great. Please add this to a new MarkDown (`.md`) file in the `doc/` folder and then submit a PR

mimaxwell, 2018-03-26:
https://github.com/cabeese/crab-tracker/pull/44

cabeese, 2018-04-06:
Closed by #44.

closed 2018-04-06 by cabeese